ostat/gridfinity_extended_openscad #126: The label style "pred - labels by pred" is no different from the "normal" label style

Semi0ne, 2025-02-23:
In the Parametric Model Maker I tried to add a "pred - labels by pred" label to a simple bin. Visually there is no difference to a "normal" label.

I exported two simple bins, one with a "pred" label, the other with a "normal" label. I then imported them into Bambu Studio and once again there is no visual difference.

I did not print them out because I'm about to watch a movie...I could print them out tomorrow if you want to be absolutely sure it's not some visual bug (seems unlikely).

zwamdurkel, 2025-09-02:
I noticed this too. I'm new to gridfinity extended and I had no clue what all the different label options meant. By now I've figured out some of it, but it took me a chunk of time. I'm still not sure what the pros and cons are when comparing i.e. cullenect and pred. It would be nice if there was some documentation on when one might use each label style. Unfortunately this issue seems to be neglected.

Kira02420, 2025-11-14:
Im currently having the same problem. Would be really nice if it got patched

closed 2025-12-28 by ostat