ostat/gridfinity_extended_openscad #128: Container Version - fork from drawer maybe

andybellgbc, 2025-02-27:
I posted this on Maker World and am posting here after rereading the description.

This is an excellent resource and covers SO many options! Thank you.

Could you make a 'container' version? This is similar to the drawer (and can almost be made with the drawer tool). It would be a container for gridfinity bins. The drawer could be used as a container by itself now, but does not interlock for stacking and lacks a few options from the cabinet and bins that could be integrated.

Adding a container as a separate file would add primarily add and option for a stackable base. But as a separate item could then have the wall pattern options and the cutouts options (from the standard bin). Could also have feet/bumper option.

And (since I am dreaming) a label space or editable text space on the face/front, would be nice.

andybellgbc, 2025-03-04:
I took the stackable Gridfinity baseplate by Jufy111 (which prints well, but is lacking sides and half grid support) and sliced it up and added it to a sample 2x2 from your drawer maker, It printed and is very strong but lack some of the supporting chamfers had some stringing unsupported edges. I can probably brute force my way into the 5x5 I am hoping for eventually but also feel like it is something others would find useful.  I am attaching a couple renders and pictures.

[image]

[image]

[image]

andybellgbc, 2025-03-17:
I posted a comment over in 131 with a couple of pictures of something that I was working on - my way of working out ideas and learning some CAD modeling.

While the items pictured above work (I printed several in 5x5) I was really hoping for something that was more milk crate like -  stackable and held the stacked items. But was more than just an oversized bin (nesting doll) which I posted my test prints in #131.

So I was thinking about a different route... And I think this would work (am printing a pair of 1x1 tests now) 

The foot is recessed into the crate; as such the bins sit above the lip of the crate. (Conceptually if you didn't care to have the crate above lock the bins below you print undersized by 1u bins.) Added ribbing to look more rugged and to effectively handle a stack. This is 1x1x6 my testing size (I plan for primarily 5x5s) the walls and ribs are all 2mm

What do you think? Is this a decent idea? I have not found any like it in my searches.

[image]
[image]

andybellgbc, 2025-03-25:
Here is what I have come up with 

https://makerworld.com/en/models/1246007-cratefinity-stacking-container-for-gridfinity-bins#profileId-1267471

Includes a 1x1 sample set

I have run about 8 5x5 half 6u tall, half 4u and a couple 2x4 6u. 

I'm sure there is room for improvement /options but beyond my abilities.

ostat, 2025-06-26:
I have made some progress on this but have not pushed it here yet. 

I was able to use the drawer from the chest as a base for the container. 
I have added some of the cup features including wall pattern and cutouts.  

I also want to add some hook options. 

I still need to clean up so it can be shared.