ostat/gridfinity_extended_openscad #129: Sliding lid handle

benz145, 2025-02-27:
I love the sliding lid option! When using the 'Not Stackable' option, I don't see an explicit way to give it a little handle to make it easier to grip when sliding out. Using the existing options, I printed the lid with a lip and the cup without a lip, resulting in a perfect little handle. 

[image]

Maybe something like this can be a built in option?

ostat, 2025-02-27:
Interesting idea. Also nice solution. 

I think we should add a couple of options, one for a pull like you have added, and maybe a slim line finger mail pull, or a hole for a pull. 

I saw your remix on makerworld, it's a great looking bin. Thanks for sharing and linking to my model. The holes look shallower than normal. What size did you go with and how well do they hold?

benz145, 2025-02-27:
All great ideas! I thought about using the lid cut-out to do a finger pull, but worried that the bits could fall out. It would be great to have multiple options for multiple use-cases.

I'm pretty sure I used the default settings for hole height and box height as a test, and it just happened to be the perfect height such that the lid is less than 1mm from the tip of the bits, so they have no chance of falling out of their holes when the box is flipped. I expanded the hole sizes by 0.1mm if I recall, to have a little bit more tolerance for the bits, and it feels just right for the bits I have.

I'm not sure I fully understand the question about the lid recess. I used default options as much as I could, but I changed the wall thickness to 1.6mm for some extra strength (and I made sure to do the same in the lid generator). 

Both pieces feel like they fit together perfectly except there is maybe a 0.5–1mm gap between the lid's 'handle' and the edge of the cup. 

[image]

Here's a few more pictures in case they are helpful:

[image]
[image]
[image]

wjcarpenter, 2026-02-16:
FWIW, I find the lid lip a little fragile to work as a handle. If there is any stickiness to the sliding action, it puts a lot of stress on that right angle. For the non-stackable option, a fillet on the joint would add some strength. 

Another out-of-the-box option would be to give up some space in the cup for a narrow label, say 3 or 4mm instead of the default 14mm. Then do a cutout on the lid that exposes that label (as is being discussed in issue #132). That little height difference gives a good fingernail grip to pull the lid out.