ostat/gridfinity_extended_openscad #130: Extension tabs misplaced when using efficient floor

johndoe0815, 2025-03-04:
The extension tabs start too much at the bottom, going into the actual ground plate. Therefore the two pieces did not fit properly together. No matter how I adjust the settings, it always starts from the bottom with the first tab. This makes it also impossible to e.g. have only one big tab per side centered.
This does however only happen with efficient floor activated. Without efficient floor it looks okay.
[image]

Secondly, a feature request instead of bug: Enable extension tabs also for the ground plate, not only for the sides.

ruby232, 2025-03-22:
The problem is in this part, but I have not yet found the issue
```
 floorHeight = calculateFloorHeight(
        magnet_depth=cupBase_settings[iCupBase_MagnetSize][iCylinderDimension_Height], 
        screw_depth=cupBase_settings[iCupBase_ScrewSize][iCylinderDimension_Height], 
        center_magnet=cupBase_settings[iCupBase_CenterMagnetSize][iCylinderDimension_Height], 
        floor_thickness=floor_thickness,
        filled_in="disabled",
        efficient_floor=cupBase_settings[iCupBase_EfficientFloor], 
        flat_base=cupBase_settings[iCupBase_FlatBase]) +
        calculateCavityFloorRadius(cupBase_settings[iCupBase_CavityFloorRadius], wall_thickness,efficient_floor)-tabThickness;
```
When **efficient_floor** is **on**, **floorHeight** is -0.7.
I forced **floorHeight**  to 5 ,and it worked for me.

ruby232, 2025-03-29:
I have the similar issue in vertical divider and efficient_floor on. Those features are compatibility together?

ostat, 2025-03-29:
It should work, but I think I have a bug in the lower z calculation.

ostat, 2025-11-08:
Hi, sorry for the delay. 

This should be fixed now.

[image]