ostat/gridfinity_extended_openscad #131: Feature request : Option to Expand Cup Walls for Internal Stacking

movrack, 2025-03-06:
Hello,  
First, thank you for your amazing work on t

he Gridfinity Extended OpenSCAD project.  
I’m submitting this feature request following discussions in the comments on [MakerWorld](https://makerworld.com/fr/models/481168-gridfinity-extended-official?from=search#profileId-554234).  

**Requested Feature**:  
I’d like to see a new option that allows the walls of the cups to be expanded outward by a user-defined distance (e.g., x mm) at each level or compute based on the level of inner Stacking and wall tickness. This would enable internal stacking of cups, similar to Russian nesting dolls (matriochka), where one cup fits inside another and not anymore on classical over bottom Stacking.

**Objective**:  
This feature would serve two main purposes:  
- Allow multiple cups to be stacked internally for compact storage when not in use.  
- Enable the use of smaller cups inside a larger one to create internal separations, treating the outer cup like a grid with walls.  

**Use Case**:  
My specific use case is to design a drainer system. The outer cup would act as a sieve, while smaller cups inside would hold items to be drained, such as brushes, straw brushes, soap, sponges, etc. This would keep everything organized while allowing proper drainage.  

**Changes to apply**:  
- The walls of the cups should shift outward by x MM ( or depending on wall tickness and the level of outer stack) while preserving the internal grid size (to maintain compatibility with the Gridfinity system).  
- The base of each cup should be widened to stay connected to the expanded walls.  
- If internal walls are used, they should be extended to match the new outer wall positions.  

**Impact on Existing Features**:  
- This feature would disable classical stacking, as the cups would now stack internally. It would be an exclusive mode: either internal stacking or classical stacking, but not both.  

**Additional Notes**:  
I’m aware that modifying the pitch alters the grid size, but my goal is to keep the internal grid dimensions unchanged. Only the walls should move outward, ensuring the cups remain compatible with the standard Gridfinity system while adding this new functionality.  

Thank you for considering this request—I believe it would greatly enhance the flexibility of this project

ostat, 2025-03-10:
I believe this is the same as request #128.
Possibly not the  exact same, but both should be able to be coved by the same solution.

andybellgbc, 2025-03-17:
I agree and thanks for keeping me in mind. Doing some cut and paste in the slicer as well as some work in onshape - I have something which closely resembles this. These are 1x1x6 test items 

[image]
[image]