ostat/gridfinity_extended_openscad #133: bug: removable divider walls broken when using custom wall tickness

movrack, 2025-03-10:
I made a try of wall tickness to 2,5.
Added custom divider, and then it appear that some side doesn't have the lines to attach the deviders.
See picture

[image]

ostat, 2025-03-10:
I tried to duplicate, however was not able to reproduce the issues. I tried both locally in OpenSCAD and MakerWorld.

In your image I see two Issues. One is that one wall has divider slots that are not attached to the wall. The second issue is one wall has no slots at all.

It looks like you are using the makerworld customizer? Can you retry and confirm if its still happening? Is there a specific setting that is causing the issue?

[image]

ostat, 2026-01-16:
This is fixed, you can try it here
model https://makerworld.com/en/models/1883145-gridfinity-extended-removable-divider-walls#profileId-2017083