ostat/gridfinity_extended_openscad #136: How are the drawer chests recommended to be printed?

benz145, 2025-03-12:
I see the default rendering orientation of the drawer is upright (the same way you'd use it after printing), but considering all the support that would be needed I'm doubting this is the ideal print orientation. 

[image]

Are there any docs/recommendations about the best way to print the drawers? Also is there a better / more preferred forum for these kinds of questions?

ostat, 2025-03-12:
I would print it on its back. 
Depending on what features you enable you will still need supports.

Monniasza, 2025-03-13:
@benz145 You can enable supportless feet on both top and botton to print the chest on its back without the need for supports.