ostat/gridfinity_extended_openscad #138: Feature Request: Partial lip edges when using non-42mm multiples

56KBs, 2025-03-18:
When creating containers using a custom size that is not a division of 42mm it'd be great if I could set side(s) of the container to not contain a lip, to allow stacking with full size containers.

For example, I would like to generate a 2 x 2.5 container and use two to fill a 2 x 5 space in the grid, due to the lip existing on the edge of the 0.5 section of the container, I can only stack containers that also use the 0.5 size. If I could disable the lip on the front/back/left/right I could still stack containers on top.

ostat, 2025-03-29:
Good idea.

ostat, 2025-05-04:
I have added an option to the lip called `Non Blocking`  8637d74

[image]

closed 2025-08-15 by ostat