ostat/gridfinity_extended_openscad #143: Add slots to removable walls

mmalmeida, 2025-04-06:
The removable walls addition is great!

How hard would it be to add the slits to the removable walls themselves too? Ideally with an option to have the slits on one side or both its sides)

This (in addition to allowing the removable walls' length to be configurable, but that can be worked around by creating different sized bins)  would allow you to configure inner bins with huge flexibility - bigger divisions, smaller ones inside it, etc - pretty much like the fixed inner bins section!

ostat, 2025-05-04:
To clarify, are you thinking have slits in the wall that go half height? and the wall interlock and are full length? 
Or having the wall have slots only on the side, and then smaller walls could be used to bridge from the wall to the next wall or bin wall.

mmalmeida, 2025-05-04:
The latter.

I know this might add a difficulty in printing these smaller walls - not really sure how we'd choose the exact size (maybe in cm ?), but it would make this extremely customizable

ostat, 2025-05-04:
The size would be based on the slot spacing. 
The slot spacing on the walls would need to match the slot spacing on the dividers, so this would dictate the size of the smaller dividers. Similar to the bent wall dividers. 

I think a working solution should be achievable without too much added complexity. Will try get to it...

ostat, 2026-01-16:
this is done, you can try it here
model https://makerworld.com/en/models/1883145-gridfinity-extended-removable-divider-walls#profileId-2017083