ostat/gridfinity_extended_openscad #151: Cullen v2 label on front of bins?

jordanrinke, 2025-04-28:
Any thoughts on adding an option to put a Cullen v2 label on the front of the bins? I know it would push the rounding back a bit and push the wall in a touch (might be a non-issue with the finger slide). Curious if anyone has thoughts or thinks it is a good idea. If people are into it, I might see if I can hack it together.

Thinking push the mount back so it sits flush with the bin front at the bottom of the bin, so it mostly takes up space where the finger slide is giving it up and on bins larger than 1 unit wide, do the same as the current top implementation and span the face of all the bins.

Haven't modeled it up but kicking around the idea, let me know if there is interest or if folks have better thoughts/ideas for labeling the fronts of bins.

ostat, 2025-05-16:
This looks great.

There is some other label work happening. So this is going to fit in well.

ostat, 2025-05-17:
Might be good to have a recessed and non recessed option.

jordanrinke, 2025-05-18:
Hadn't considered a non-recessed option but I re-used the existing label socket stuff as much as possible so it should be an easy config to adjust the offset and add a small box around it. Will try and get a PR up in the next week or two. I also fixed efficient floor to support a center magnet, and added an option to specify which of the 4 corner magnets are rendered (and that supports efficient floors as well) since for my own personal use I am only rendering the front right magnet.

ostat, 2025-05-18:
Sounds great. 

If you can try to push the different changes in a different PRs. 
This makes it easier to review and keeps the changes moving faster. 
So ideally one PR for efficient floor fix, one for the magnet change and so on.

jordanrinke, 2025-05-19:
Oh for sure, each is its own branch and I have then all applied to my local main, I've just not gotten around to cleaning them all up for individual use and submitting PRs. They all work but they aren't clean currently.