ostat/gridfinity_extended_openscad #171: Multi-cylinder support

theontho, 2025-06-15:
With camera lenses and camera filters, everything is a form of a cylinder.  A parametric generator for cylinders would be a nice add on to this system.

There would be 3 or 4 main kinds of cylinder types to support:

* Rows of cross cylinder cutouts where you can specify diameter: https://makerworld.com/en/models/862998-lens-filter-holder-gridfinity-1x2x3?from=search
* Stacked vertical and horizontal cylinder cross cutouts, you specify a list of diameters, height and chamfers and they get cut out in order: 
  * horiztonal: https://makerworld.com/en/models/1391402-nikon-z-24-120-f4-s-case-gridfinity-no-filter?from=search#profileId-1441689
  * vertical: https://makerworld.com/en/models/1488719-gridfinity-sony-e-mount-lenses
* Basic vertical, cross column and cross row cylinder cutouts like gridfinity rebuilt https://gridfinity.perplexinglabs.com/pr/gridfinity-rebuilt/0/0
* Minimalistic "base & cylinder" holders: https://makerworld.com/en/models/791460-gridfinity-camera-lens-storage-bins?from=search#profileId-729936

trysten, 2025-08-13:
Since this doesn't relate directly to gridfinity (extended), I think this issue should be closed.

If you refine your ideas and make some sketches of what you're imagining you may send them to me and I might make a new project for your idea.

ostat, 2025-08-13:
There was a reddit post that was related to this.
https://www.reddit.com/r/gridfinity/comments/1mc8wug/customizable_camera_lens_stencil/

I commented that adding this as a dedicated script would be a cool idea if they were interested. They indicated that they would send it through.