ostat/gridfinity_extended_openscad #173: Feature Request: Stackable Item Holders

jonaskello, 2025-06-16:
So we eg. can have stackable battery holders like these:

https://makerworld.com/en/models/678776-gridfinity-aa-battery-storage

ostat, 2025-06-17:
This is supported, but its not obvious unfortunate.

There is an option called **Auto Bin Height** that sets the bin height based on the the hole depth. Turn this off to use use the z height value you mentioned.

You can also change the hole depth to not use the the value in the pre configured options available.
Let me know how you go.

[image]

jonaskello, 2025-06-18:
This option does exactly what I need, not sure how I missed that.

Tried it and it worked perfectly, thanks for pointing it out and for creating this amazing project :-).

closed 2025-06-18 by jonaskello