ostat/gridfinity_extended_openscad #178: Setting a wall thickness higher than 3.74 fills the whole cup

SiJiL82, 2025-07-19:
`gridfinity_basic_cup.scad`
Increasing the Wall Thickness parameter over 3.74 fills the whole cup solid.

Wall thickness set to 3.74:
[image]
Any higher value:
[image]

This doesn't seem to be affected by any other settings as far as I can tell (as in you can't change another setting and then select a higher wall value and have it work).

ostat, 2025-07-19:
Hi, nice catch. 

I suspect there is a bug in the math, the internal corner radius is going negative. 

I will take a look.

ostat, 2025-07-21:
This has been fixed in 92b14870

closed 2025-07-21 by ostat