ostat/gridfinity_extended_openscad #208: Error when enabling "wallpattern dividers enabled"

thepauly, 2025-09-21:
I'm getting this error in OpenSCAD and a similar one in the Makerworld customizer when trying to enable a wallpattern pattern on dividers.

ERROR: Assertion '(is_num(canvasSize.x) && (canvasSize.x > 0))' failed: "canvasSize.x must be a positive number" in file modules/module_patterns.scad, line 329

[image]

mrene, 2025-09-22:
I still get the error with `gridfinity_basic_cup` using 5f458b197dd6f7ad344cf1892ef855cef003cf75

ostat, 2025-09-22:
I see, try now?

thepauly, 2025-09-22:
Thanks so much.

Error has gone but there is no pattern on the dividers.

[image]